4am-robotics/cob_driver #207: base_drive_chain's diagnostics message is published with 100 Hz

ipa-bnm, 2015-03-03:
The base_drive_chain diagnostics message is published with 100Hz. This is not necessary and should be reduced. I think every half second or second is absolutely enough and could be easily done with a simple ros timer.
@ipa-mig What do you think?

ipa-mig, 2015-03-03:
yes, this is fine for me.
We need to take into account, however, that there are two diagnostic msgs. The global one does not need to go out at 100Hz, as far as I now. @ipa-fmw @abubeck can you confirm this?
The local diagnostics reside in the base_controller namespace and are currently misused as a means of communication with the undercarriage_ctrl. So this should be kept at a high rate, as far as I see.

ipa-fmw, 2015-03-03:
+1 for low-frequency (1Hz) global diagnostics

ipa-fxm, 2015-04-20:
change this on the Wednesday-Hackathon?

ipa-mig, 2015-04-20:
if you find the time, would be perfect!
Be careful though about the "misued" local diagonstics. I don't think they should be set down (at least not without extensive testing in combination with the undercarriage_ctrl).

ipa-fxm, 2015-04-20:
Can you give an example for one of those 
 > "misued" local diagonstics

ipa-mig, 2015-04-20:
https://github.com/ipa320/cob_driver/blob/indigo_dev/cob_base_drive_chain/ros/src/cob_base_drive_chain.cpp#L691-715
here, the diagnostics are used to communicate with the undercarriage ctrl about the initialization status of the wheels/motor controllers. This is handled here:
https://github.com/ipa320/cob_driver/blob/indigo_dev/cob_undercarriage_ctrl/ros/src/cob_undercarriage_ctrl.cpp#L347-416

Not nice. But working. Will hopefully be changed (or deprecated, if we get rid of the base_drive_chain completely) in the process of switching to ros_control/cob_control for the old drives as well. But requires significantly more work.

@ipa-fxm

ipa-bnm, 2015-05-26:
http://wiki.ros.org/diagnostic_updater

closed 2017-03-16 by ipa-fxm