4am-robotics/cob_driver #224: Critical compiler warnings in phidgets and lights

ipa-bnm, 2015-05-26:
This compiler warnings should get fixed:
- cob_phigets
  - [implicit conversion from 'double' to 'unsigned int' changes value from 0.5 to 0](https://github.com/ipa320/cob_driver/blob/2562b9247de5bc99141a3737b566360abda0b822/cob_phidgets/ros/src/phidget_manager.cpp#L80)

- cob_light
  - [control reaches end of non-void function (no return value)](https://github.com/ipa320/cob_driver/blob/2562b9247de5bc99141a3737b566360abda0b822/cob_light/common/src/serialIO.cpp#L167)
  - [control reaches end of non-void function (no return value)](https://github.com/ipa320/cob_driver/blob/2562b9247de5bc99141a3737b566360abda0b822/cob_light/common/src/serialIO.cpp#L172)

ipa-bnm, 2015-05-26:
@ipa-bnm

ipa-fxm, 2015-05-27:
Also have a look at these (minor) catkin_lint hints:
```
cob_phidgets: CMakeLists.txt(8): notice: line is not indented properly
cob_phidgets: CMakeLists.txt(14): notice: extra arguments in else()
cob_phidgets: CMakeLists.txt(16): notice: extra arguments in endif()
```

ipa-fxm, 2015-09-04:
@ipa-bnm 
any progress here?

ipa-fxm, 2017-02-09:
forget about `catkin_lint`
done in #314

PR #314 (2017-02-09): Compiler fixes
@ipa-bnm 
fixes #224
closed 2017-02-09 by ipa-bnm