4am-robotics/cob_driver #236: [cob_light] Handling `sim_enabled` param in simulation

ipa-fxm, 2015-08-18:
See my comments about "Warning B" in https://github.com/ipa320/cob_simulation/issues/87#issuecomment-132086878

So far, we are using the same yaml for simulation and real hardware...and in that yaml `sim_enabled` is set to false...maybe we should make it a private parameter that can be set via launch file argument in https://github.com/ipa320/cob_robots/blob/indigo_dev/cob_bringup/drivers/light.launch?

ipa-fxm, 2015-08-18:
As `sim_enabled already is a private parameter....https://github.com/ipa320/cob_robots/pull/326 should fix the warning about failed serial connection in simulation

ipa-bnm, 2015-08-18:
as already in ipa320/cob_robots#326 commented, this is ok!

closed 2015-08-25 by ipa-nhg