4am-robotics/cob_driver #249: cob_scan_unifier is not merging complete scans in simulation

ipa-bnm, 2015-11-19:
_From @ipa-bnm on September 17, 2015 12:5_

scan unifier produce gap on the back of cob4-2
[image]

@ipa-frm @ipa-mig 

_Copied from original issue: ipa320/cob_navigation#53_

ipa-bnm, 2015-11-19:
_From @ipa-mig on September 18, 2015 6:1_

hmm, also, there is a z-offset (which will probably not be a problem for our use-cases... but we should change that anyways...

ipa-bnm, 2015-11-19:
the z-offset seems to be ok. Imagine the scanners are mounted in different z positions (eg. our BMW). So transform the scans to the ground is ok, from my opinion. A better solution would be the usage of a point cloud instead a merged scan message.

ipa-bnm, 2015-11-19:
https://github.com/iralabdisco/ira_laser_tools 
there you can find a tool with the same purpose as cob_scan_unifier.
Difference: 
- no gaps
- dynamic reconfigure
- offers point cloud 2 as well

Perhaps we should stick to this package? We could even copy the code from this package to our repo, because it has no license at all.

destogl, 2016-01-08:
@ipa-bnm yesterday I ported cob_scan_unifier to tf2, are you proposing to use this other repository and let cob_scan_unifier to die out? This is important to me to know in which direction to go.

Regarding this not merging scans, I also saw it with real scans but can not really say where is the problem.

ipa-bnm, 2016-01-12:
@destogl for now we will stick to cob_scan_unifier. For the moment there are no plans to switch to another solution. But if we continue to have problems with cob_scan_unifier , it could be that we change or reimplement it. I will also have a look at your PR

ipa-mig, 2017-03-09:
Also, check the angle increment, see [comment](https://github.com/ipa320/cob_driver/pull/324#discussion_r105103344)

ipa-fxm, 2017-03-20:
@ipa-mig, is @mig-em working on this after we merged #324?

ipa-mig, 2017-03-20:
he should be :-)

mig-em, 2017-03-20:
I can't reproduce this bug in the current scan_unifier. I tested in a corner of ipa-apartment and farer away to increase the effect, but no gaps showed up.

ipa-fxm, 2017-03-30:
@ipa-bnm @ipa-mig 
Have you had a chance to test the latest version of `cob_scan_unifier` (using synchronizer)?
So has this issue become obsolete?

fmw-hb, 2017-04-18:
this occured today: 
[image]
is this related to the issue? (latest commit is from April 5th 2017)