4am-robotics/cob_driver #268: [cob_scan_unifier] LookupError on start up

ipa-fxm, 2016-03-04:
Appears once on startup, probably simply requires a sleep after tf::TransformListener initialization (for filling the buffer):

```
Node: /scan_unifier
Time: 16:05:39.612337463 (2016-03-04)
Severity: Error
Published Topics: /rosout, /scan_unified

Lookup would require extrapolation into the past.  Requested time 1457103936.557201524 but the earliest data is at time 1457103937.074180613, when looking up transform from frame [base_laser_front_link] to frame [base_link]

Location:
/u/fxm/git/care-o-bot/src/cob_driver/cob_scan_unifier/src/scan_unifier_node.cpp:LaserScan scan_unifier_node::unifieLaserScans:252
```

Also, `unifieLaserScans` should be `unifyLaserScans` :wink:

ipa-nhg, 2016-04-22:
I had the same issue during the cob4-5 setup. But in this case the error appears constantly and not only  just once on startup. Checking the visualization (RVIZ), I notified that the errors loggin corresponded to a wrong transformation/scanner lecture. The driver mixed the header and the data of the messages, for example the data read by the scan_front were transformed to the scan_right frame. After set the loop_rate of the driver to 10Hz, it is working properly. @ipa-bnm any ideas?

ipa-mig, 2017-02-08:
@ipa-nhg do you remember why you had to set the loop rate to 10 Hz here?

ipa-fxm, 2017-03-16:
fixed in https://github.com/ipa320/cob_driver/pull/324

closed 2017-03-16 by ipa-fxm