4am-robotics/cob_driver #276: Lidar scan is warped

mehditlili, 2016-04-20:
I had some issues with two sick S300 Professional on my robot and could not get the scans to overlap even after calibrating the lidars realtive to robot center. As it was not caused by some offsets in the lidars pose, I started checking the what the lidar driver was delivering. I also found another [S300 driver](https://github.com/bohlender/sicks300) that delivered a slightly different scan while observing the same thing. This motivated me to look more into this problem. I found out that the scans I get using your driver have an angle increment of slightly more than 0.5 degrees and is computed from angle_start, angle_stop and size of the scan array, while the angle increment in the other driver I tested was explicitly set to 0.5, number that is stated in the data sheet of the Sick S300.
In order to get my IPA Sick driver to deliver a similar signal to the other driver, I had to change [this file](https://github.com/ipa320/cob_driver/blob/indigo_dev/cob_sick_s300/common/src/ScannerSickS300.cpp) and set the "- 1" in line 264 to "+ 4". I am still not sure about the reason why it works this way but now the increment computed is equal to 0.5 degrees and the extrinsic calibration works and the front and rear laser are overlapping. Can you please take a look into this?

ipa-mig, 2016-04-20:
@mehditlili thank you for providing this issue.

Given your description, this seems to be a duplicate (of some sort) of #256.
I guess that your fix is actually not correct but will probably serve the purpose for now.

(FYI: you can link directly to lines in GitHub by clicking on the line numbers in the respective files or by appending `<URL>#LXXX-LYYY` to the URL)

mehditlili, 2016-04-20:
I am not sure about the protocol, I just started the launch file until now and it starts publishing the laser scans.

ipa-mig, 2016-04-21:
Can you please post the product number of the Scanner? If it is `S30B-2011DB` you have the old Professional CMS with protocoll `0x0102`, if it is `S30B-2011GB` you have the scanners with the new protocoll `0x0103`.
You could also check the firmware of the scanners itself. the old one has 1.4x, if I'm not mistaken, the new one 2.x.

mehditlili, 2016-04-21:
We use the [S30B-2011DA](https://www.sick.com/media/pdf/3/43/843/dataSheet_S30B-2011DA_1026822_en.pdf)

mehditlili, 2016-04-29:
What would stand against replacing [this line](https://github.com/ipa320/cob_driver/blob/indigo_dev/cob_sick_s300/common/src/ScannerSickS300.cpp#L264)  with 
`dAngleStep = 0.5*180.0/ScannerSickS300::c_dPi;`  ?
0.5 is given as angle resolution in the datasheet of the Sick LMS and seems to be constant for all models.

ipa-mig, 2016-04-29:
Well, true, but you can configure the output to be something different (but why would you want to do that?), and if you don't read enough data to put on the ranges, you'll really be in trouble.
But you are right, this needs to be adapted properly. We could also add a sanity check based on this...

PR #317 (2017-03-16): [cob_sick_s300] fix bug reading not all scan points
This PR fixes #256 and fixes #276 by (more) correctly implementing the size calculations acoording to the (updated) Sick telegram listing.

Tested with:
- [x] S300 Pro CMS (old protocol)
- [x] S300 Expert (compatability mode)
- [x] S300 Expert (new protocol)

This PR also comments out a warning about "old" parameter sets, as those parameter sets actually are buggy --> #318

@ipa-josh
closed 2017-03-16 by ipa-fxm