4am-robotics/cob_driver #293: Timestamp calculation is not implemented

mingless, 2016-09-04:
In ScannerSickS300.cpp `getScan()` function always returns `	iTimeNow=0;`
and does not assign value to iTimestamp. This is used by cob_sick_s300.cpp in `publishLaserScan()` to calculate the message timestamp.
``` c++
			// Sync handling: find out exact scan time by using the syncTime-syncStamp pair:
			// Timestamp: "This counter is internally incremented at each scan, i.e. every 40 ms (S300)"
			if(iSickNow != 0) {
				syncedROSTime = ros::Time::now() - ros::Duration(scan_cycle_time);
				syncedSICKStamp = iSickNow;
				syncedTimeReady = true;

				ROS_DEBUG("Got iSickNow, store sync-stamp: %d", syncedSICKStamp);
			} else syncedTimeReady = false;
```

Since it's not implemented it only obfuscates the code while always ending up with `ros::Time::now()`.

Will the timestamp calculation be implemented?

ipa-fmw, 2016-09-09:
@ipa-josh: can you please comment on this?

mingless, 2016-11-09:
Since I am not currently using any Sick scanners it's not a pressing issue for me, but if needed I am available and will gladly contribute.

ipa-mig, 2016-11-10:
@mingless thank you for the offer. If you find the time, we are happy to accept any contributions.

Otherwise, we ahve it on our list and will hopefully find the time soon to implement this. @ipa-josh