4am-robotics/cob_driver #318: [cob_sick_s300] fix fields parameter and handling.

ipa-mig, 2017-02-13:
Currently, there is a warning about using the `fields` parameter, as this is the way to go.
```
"No params for the Sick S300 fieldset were specified --> will using default, but it's deprecated now, please adjust parameters!!!"
```

However, this is not properly implemented. The driver only supports one field, which defeates the purpose of this parameter. Also, this parameter is nowhere documented.

ipa-fxm, 2017-03-09:
Can someone post what the old config looked like (and where it needs to be set) and what the new config looks like (and where it needs to be set)?

Also, I'd appreciate a few detailed bullet points about what needs to be done implementation-wise to fully implement this feature (and where)

ipa-mig, 2017-03-09:
@ipa-fxm see edit above.