4am-robotics/cob_driver #327: header files cob_omni_drive_controller

andreeatulbure, 2017-03-09:
Hello,

I tried to extend the cob_omni_drive_controller in order to add a new feature for the raw3-5. I noticed that there are some header files in the src directory of the library (GeomController.h and WheelControllerBase.h). Is there any reason why they are not in the include directory?

Thanks,
Andreea

ipa-mdl, 2017-03-09:
> Is there any reason why they are not in the include directory?

The controller headers are for code reuse only, they are not part of the public API.

ipa-fxm, 2017-03-16:
@andreeatulbure 
is this issue still relevant? if yes, please open it in cob_control (this is where the `cob_omni_drive_controller` package is located)
Closing!

closed 2017-03-16 by ipa-fxm