4am-robotics/cob_driver #338: [cob_sick_s300] frame rate low with RS-422

ehocquard, 2017-06-02:
Hello everyone,

I am working with sick s300 advanced sensors for a UAV project. My system is designed with ROS and thanks to your work I have compatibility with sick and ROS.

The sick is connected via a RS-422 interface set up at 500k baud on continuous data transfer, it works.

However, I do observe a rate of scan transfer (on the topic) of **12 Hz** only.
I was wondering if anyone had similar results, it sounds a bit low rate  for a scanner that will be use for autonomous driving. Is this limited by the RS-422 or there is another bottleneck that I don't see?

If anyone can share experience on that I'll be grateful!

Best regards,
Etienne H.

PS: sorry if raising an issue for this is not appropriate!

ipa-fxm, 2017-06-02:
there is a parameter `publish_frequency` which defaults to 12 :wink: 
https://github.com/ipa320/cob_driver/blob/indigo_dev/cob_sick_s300/ros/src/cob_sick_s300.cpp#L146

ehocquard, 2017-06-06:
Indeed, thank you for pointing it out.

However changing the parameter does not change the rate of the received message, hence my question/issue on similar experience and maximum speed reached by others.

Log with a publish frequency set to 50:
"average rate: 12.415
	min: 0.028s max: 2.572s std dev: 0.06747s window: 1366
average rate: 12.417
	min: 0.028s max: 2.572s std dev: 0.06715s window: 1379
average rate: 12.420
	min: 0.028s max: 2.572s std dev: 0.06683s window: 1392
average rate: 12.422
	min: 0.028s max: 2.572s std dev: 0.06655s window: 1404
average rate: 12.424
	min: 0.028s max: 2.572s std dev: 0.06624s window: 1417
average rate: 12.427
	min: 0.028s max: 2.572s std dev: 0.06594s window: 1430
average rate: 12.429
	min: 0.028s max: 2.572s std dev: 0.06567s window: 1442
average rate: 12.431
	min: 0.028s max: 2.572s std dev: 0.06537s window: 1455
^Caverage rate: 12.433
	min: 0.028s max: 2.572s std dev: 0.06513s window: 1466"

ipa-mig, 2017-06-26:
Hi @ehocquard 

There are several things to consider here:
1. The scan cycle time of the S300 is 40ms, i.e. the maximum frequency is actual 25Hz. Thus, setting `publish_frequency` to anything above will not do any good :-)
1. I'm actually not sure, if the parameter mentionend by @ipa-fxm actually does what it is supposed to :-) The only thing it can do (if it does) is to limit the publish frequency
1. The configuration of the scanner has a high influence
    1. 500kbaud is a must, otherwise, the rate will drop significantly
    1. If you specify "measurement ranges", i.e. limit the output angles `angle_min`/`angle_max` in ROS terms (even just one with the full range), the frequency is halved
    1. If you configure reflector marker output, the frequency is halved (I don't think this is cumulative, though).
    1. If you output I/O-data, there might be a drop in frequency as well
1. Last but not least, in my experience, there is a difference between the versions of the S300. The old S300 Pro/Expert CMS I could not get above 12.5Hz, the new ones can actually reach the 25Hz, if configured correctly

I hope this clears some things up :-)

I'll close this issue. But feel free to reopen if you have additional questions...

closed 2017-06-26 by ipa-mig

ehocquard, 2017-06-26:
Hi @ipa-mig ,

Thank you very much for the detailed reply.
I do agree with your comments.

In addition, I have done some analysis from checking baudrate with a scope to code analysis and packet transfer rates. I confirm that with the old S300 (the one I have), 12.5 Hz is the maximum rate we can get via RS-422.

Again thank you to those who developed the package!