4am-robotics/cob_driver #340: [cob_mimic] consider c++ implementation

ipa-fxm, 2017-07-18:
motivated by discussion https://github.com/ipa320/care-o-bot/issues/42#issuecomment-316046500

The ROS release system does not support `python-pip` dependencies...thus, we cannot release `cob_mimic` with a dependency on `python-vlc-pip`.
We should implement `cob_mimic` in C++ using [`libvlc-dev`](https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml#L2987-L2991) which is provided by `apt` which is supported by the ROS release system.

ipa-fmw, 2017-07-19:
@ipa-bnm is investigating what would need to be changed

ipa-bnm, 2017-07-21:
It should not be a very big task to port the python code to c++. I think half a day should be enough!
I can do it!

ipa-fmw, 2017-07-21:
this is currently blocking the indigo/kinetic release. 
@ipa-bnm please do.

PR #344 (2017-07-24): [mimic] ported mimic from python to c++
 - fixes #340
closed 2017-07-24 by ipa-fxm