4am-robotics/cob_driver #355: [cob_mimic] the cpp implementation is not working

ipa-nhg, 2017-08-22:
the resolution and rotation of the mimic is wrong and can not be configured using the cpp library for vlc

ipa-nhg, 2017-08-22:
could we readd the python?? node https://github.com/ipa320/cob_driver/pull/356

ipa-fxm, 2017-09-04:
For which robot has this (resolution and rotation) been wrong?
We tested the cpp-driver with various robots...and we did not experience this problem...

Also, I think we should at least use the python-drver which used the vlc-API...because that way we resolved the flickering problem as described in https://github.com/ipa320/msh/issues/714#issuecomment-302705555....even if this means we have to add the python-vlc module to the mimic package...as was done in https://github.com/ipa320/cob_driver/pull/339 without the last commit https://github.com/ipa320/cob_driver/pull/339/commits/2bf582d05a3c0ac3f3901f0b7f992bada034674f

ipa-nhg, 2017-09-04:
this was for t-labs and probably is the same problem that had paul-ingolstadt.
I had the same issue with the python-driver that uses the vlc-API , this is an issue with the parametrization of vlc. With the APIs we can't set the parameters that we need (specially` --vout glx`), the need of the parameter could also be a problem related to the graphic card driver, paul-stuttgart (where it is working) has a different graphic card model...

Please feel free to modify the argument `old` this was a very urgent hot fix... sorry

ipa-fxm, 2017-09-20:
a solution might be provided in https://github.com/ipa320/msh/issues/845#issuecomment-330476261

closed 2017-10-06 by ipa-fmw