4am-robotics/cob_driver #439: cob_sick_s300 does not accept parameters

bruiglesias, 2023-02-13:
It's my first time with a sick laser, I'm trying to change the port from "/dev/ttyUSB0" (default) to "/dev/ttyUSB1" and frame_id from "base_laser_link" (default) to "front_laser", but cob_sick_s300 always starts with default values. What can be wrong?

This is my launch file:

`
<?xml version="1.0"?>
<launch>

    
    <node name="laser_front_node" pkg="cob_sick_s300" type="cob_sick_s300" respawn="false" output="screen">
        <param name="port" type="string" value="/dev/ttyUSB1" />
        <param name="baud" type="double" value="500000" />
        <param name="frame_id" type="string" value="front_laser" />
        <remap from="scan" to="front_scan" />
    </node>

    
    <node name="rqt_gui" pkg="rqt_gui" type="rqt_gui"/>
    <node name="rviz" pkg="rviz" type="rviz"/>
</launch>
`

This is my console output:

[image]

fmessmer, 2023-02-20:
your parameters get uploaded into private namespace `/laser_front_node/port`
but the node reads them from relatve namespace `/port` - see https://github.com/ipa320/cob_driver/blob/kinetic_dev/cob_sick_s300/ros/src/cob_sick_s300.cpp#L89

this is probably rather a problem of the node than your launch file - but the node has been around for so long already - thus we dont dare to change this anymore...

just load your params to the correct namespace and you should be fine

fmessmer, 2023-03-13:
I'm closing this now

closed 2023-03-13 by fmessmer